jimmo/numato-mimasv2-pic-firmware #17: UART pass-through doesn't work after power cycle

toptensoftware, 2020-01-10:
Firstly, thank you for this great update to the Mimas firmware.

It seems that the UART pass-through only works after reprogramming the FPGA.  If you power cycle the board, the UART no longer works (until you upload a new design to the FPGA - then it works again)

Is this a known issue and if so, any the chance of a fix?

Brad

toptensoftware, 2020-01-11:
To confirm this wasn't a problem with my deisgn, I just re-routed the UART via a FTDI USB adapter connected to one of the GPIO headers and I can power cycle and it still works.  Definitely seems to be a problem with the firmware.

toptensoftware, 2020-01-12:
Please ignore... turns out this is probably related to running in a VM under VirtualBox.  With real linux machine it works fine.

closed 2020-01-12 by toptensoftware

mtrx93, 2020-12-14:
I have same problem, there is no need to reprogram, opening port on ACM0 somehow activates ACM1

toptensoftware, 2020-12-14:
Ah good to know.  Thanks for the info.